The-OpenROAD-Project/asap7 #3: ICG cells throw an error in library compiler

hchenji, 2021-02-13:
converting the .lib to .db using Synopsys library compiler. Only these ICG cells throw an error, the rest are successfully read in.

```
read_lib asap7sc7p5t_SEQ_RVT_TT_nldm_201020.lib
Reading '/pub/techlib/oss/asap7-openroad/asap7sc7p5t_27/LIB/NLDM/asap7sc7p5t_SEQ_RVT_TT_nldm_201020.lib' ...
Error: Line 10430, Cell 'ICGx1_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx1_ASAP7_75t_R' cell. (LBDB-23)
Error: Line 11379, Cell 'ICGx2_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx2_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 12328, Cell 'ICGx2p67DC_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx2p67DC_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 13277, Cell 'ICGx3_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx3_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 14226, Cell 'ICGx4DC_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx4DC_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 15175, Cell 'ICGx4_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx4_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 16124, Cell 'ICGx5_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx5_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 17073, Cell 'ICGx5p33DC_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx5p33DC_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 18022, Cell 'ICGx6p67DC_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx6p67DC_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 18971, Cell 'ICGx8DC_ASAP7_75t_R', pin 'GCLK', There is a missing timing arc between pins 'IQ' and
        'GCLK' in the 'ICGx8DC_ASAP7_75t_R' cell. (LBDB-23)

Error: Line 10436, Cell 'ICGx1_ASAP7_75t_R', In the 'ICGx1_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 11385, Cell 'ICGx2_ASAP7_75t_R', In the 'ICGx2_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 12334, Cell 'ICGx2p67DC_ASAP7_75t_R', In the 'ICGx2p67DC_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 13283, Cell 'ICGx3_ASAP7_75t_R', In the 'ICGx3_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 14232, Cell 'ICGx4DC_ASAP7_75t_R', In the 'ICGx4DC_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 15181, Cell 'ICGx4_ASAP7_75t_R', In the 'ICGx4_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 16130, Cell 'ICGx5_ASAP7_75t_R', In the 'ICGx5_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 17079, Cell 'ICGx5p33DC_ASAP7_75t_R', In the 'ICGx5p33DC_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 18028, Cell 'ICGx6p67DC_ASAP7_75t_R', In the 'ICGx6p67DC_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

Error: Line 18977, Cell 'ICGx8DC_ASAP7_75t_R', In the 'ICGx8DC_ASAP7_75t_R' cell, the 'IQ' noninput pin 
        cannot be used in the function. (LIBG-4)

```

vvashish, 2021-03-23:
We are aware of an issue with the integrated clock-gater (ICG) cells that affects their utilization during synthesis and place-and-route. We will try to address it in a subsequent release. However, I am unsure whether the issue that we see is also the reason behind the error in what you are trying to do.

harrisonliew, 2021-07-26:
I believe I have hacked around the SEQ NLDM LIB files to get synthesis to recognize the ICG cells. Here is what I did (based on cross-referencing with r1p5):

- Add the following statetable to each ICG cell:
```
    statetable ("CLK ENA SE", "IQ") {
      table : "L L L : - : L ,  L L H : - : H , L H L : - : H , L H H : - : H , H - - : - : N ";
    }
```
- Change the `function` in the `GCLK` pin to `state_function`:
```
    state_function: "CLK & IQ";
```

Here's a quick gzip + sed chaining to hack it:
```
gzip -cd <source_lib> | sed '/ICGx/a <statetable_text_above>' | sed '/CLK & IQ/s/function/state_function/g' | gzip > <hacked_lib>
```

closed 2023-11-02 by maliberty