The-OpenROAD-Project/asap7 #16: Request for ASAP7 Calibre Decks

baichen318, 2022-03-12:
Hi all,

I recently applied for the request of ASAP7 Calibre Decks (https://live-predictive-pdk.pantheonsite.io/download/) with my university email account.

However, I could not get a response after submitting my application for several weeks.

I wondered how I could get access to ASAP7 Calibre Decks.

Thank you very much!

Best regards,
Chen

Waxpple, 2022-03-19:
Me too. I want to access the deck too

mi-and-core, 2022-04-24:
(encrypted) ASAP7 Calibre Decks: 
https://live-predictive-pdk.pantheonsite.io/download/

1bing2, 2022-05-13:
Me too. I want to access the deck too

1bing2, 2022-05-13:
> Me too. I want to access the deck too

Have you handle it ?

tspyrou, 2022-05-13:
@vvashish can you help point people to a contact at ASU?

1bing2, 2022-05-16:
> @vvashish can you help point people to a contact at ASU?
Hello,
I wondered how I could get access to ASAP7 Calibre Decks. Have you handle it?
Thank you!

vvashish, 2022-05-30:
Hi,

Apologies for the late reply. These messages went to my spam folder and I
just saw them. Prof. Lawrence T. Clark ***@***.***) would be
the best point of contact. He will be able to help with the Calibre deck
downloads.

Regards,
Vinay Vashishtha

1bing2, 2022-05-30:
Can you give me the email of Prof. Lawrence T.I want to contact him for thr Calibre Decks! 
My email is bingli226@mail.ustc.edu.cn

vvashish, 2022-05-30:
***@***.***

idwwwoqq808, 2022-10-26:
> Can you give me the email of Prof. Lawrence T.I want to contact him for thr Calibre Decks! My email is [bingli226@mail.ustc.edu.cn](mailto:bingli226@mail.ustc.edu.cn)

I think this is the professor vvashish mentioned: https://search.asu.edu/profile/751415

closed 2023-11-02 by maliberty