The-OpenROAD-Project/asap7 #19: Question about the technology LEF file

pacificwind, 2022-05-02:
Hello, I'm trying to do APR with ASAP7 PDK.

I found the macro LEF file, but couldn't find the technology LEF file.
Where can I find the technology LEF file

maliberty, 2022-05-02:
In techlef_misc for the appropriate library

pacificwind, 2022-05-03:
Thank you! I have a further question.

When I when to the asap7/asap7sc7p5t_27/techlef_misc/,
there is asap7_tech_4x_201209.lef, there is no asap7_tech_1x_201209.lef file.

Where can I find the technology lef file for unscaled one?

[image]

maliberty, 2022-05-03:
asap7 doesn't provide a 1x tech lef as they were unable to verify it due to licensing issues.  In ORFS we generated one https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef

pacificwind, 2022-05-03:
Thank you very much! Your explanation helped me a lot!

I'd like to ask a minor question.
What is the ORFS? (I think it stands for OpenRoad Flow Script. But I couldn't fully understand what it is.)
And is the OpenRoad Flow project separate from ASAP7?

maliberty, 2022-05-03:
Yes ORFS is OpenRoad Flow Scripts.  It is a set of scripts to run yosys, openroad, and klayout to implement a design.  Asap7 is data describing a process (a PDK).  They are separate though ORFS can use asap7 as a PDK.

pacificwind, 2022-05-04:
Can I ask the difference between the unscaled (1x) version and the scaled (4x) version?

Is the width and height of the scaled-version 4-times smaller than that of the unscaled-version?

maliberty, 2022-05-04:
4X is four times larger than 1X.

pacificwind, 2022-05-04:
I see. Thank you very much!

closed 2023-02-21 by vijayank88

QuantamHD, 2023-07-25:
@maliberty This is a priority for our team.

maliberty, 2023-07-25:
We didn't previously because ASU wouldn't support it.  Now that they do little support anyways I guess it doesn't matter.  If it is important to you feel free to open a PR.  There might be little tweaks to 1x that aren't in 4x fwiw.

proppy, 2023-07-25:
@lpawelcz my understanding is that it would unblock #172 if we submitted https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef back to this repo, can you confirm?

maliberty, 2023-07-25:
What is #172 ?

proppy, 2023-07-25:
@maliberty sorry, I typed too fast (update the ref to include the full repo)

lpawelcz, 2023-07-26:
@proppy this would for sure be helpful because in my understanding this file is missing from ASAP7. I did some experiments with manually added [asap7_tech_1x_201209.lef](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/lef/asap7_tech_1x_201209.lef) and this resulted in layouts very similar to ones generated with ORFS but I was not able to fully unblock https://github.com/hdl/bazel_rules_hdl/issues/172. I'll post more info there.