The-OpenROAD-Project/asap7 #21: Question about LIB model

pacificwind, 2022-05-08:
Hi, I have a question about the type of LIB model.

When I go to the asap7/asap7sc7p5t_28/LIB/CCS directory,
there are three types of LIB files: CCS, CCSA, and CCSN.
What is the difference between those types?

The URL of the directory is as follows:
https://github.com/The-OpenROAD-Project/asap7/tree/master/asap7sc7p5t_28/LIB/CCS

maliberty, 2022-05-08:
@tspyrou do you know?

pacificwind, 2022-05-09:
It seems that CCSN means CCS-Noise.

[image]

It is in the 242-page of this manual.
https://picture.iczhiku.com/resource/eetop/shigyATqwoLuLmXv.pdf

closed 2023-11-02 by maliberty