The-OpenROAD-Project/asap7 #32: Question about the follow rails

pacificwind, 2022-07-29:
Hi, I have a question about the tcl file for the P&R (Placement and Route).

It is a simple question, but I have no idea what the follow rail is in the given 'example_innovus.tcl' and 'm2followRail.tcl'.
Can I ask what the 'follow rail' is?
And is it right that M2 is used as a follow rail in the tcl file?

maliberty, 2022-07-30:
In general follow rails (or follow pins) are stripes that go along the rows connecting to the std cell pins.  In some PDKs they will be on just m1 and in others on m1 & m2 (a way to get more metal without increasing the width).  In the two layer case they will have vias running along the pair to tie them strongly together.

pacificwind, 2022-07-30:
Thank you for your explanation!

closed 2023-02-21 by vijayank88