The-OpenROAD-Project/asap7 #42: Please add the paper (without paywall) and link to slide deck

NilsGraf, 2023-01-06:
Unfortunately, the paper you linked in the README is behind the [IEEE paywall](https://ieeexplore.ieee.org/document/8203889). Could you please add the paper to github for people who don't want to pay $33 to view this paper?   Also, perhaps add a link to the slide deck of the conference talk:
https://pdfs.semanticscholar.org/90db/4b674407770924c8208011a9dc4176e37a3a.pdf   Thank you!

vijayank88, 2023-01-06:
@maliberty Can we upload above PDF?

maliberty, 2023-01-08:
Not unless you have access outside of the IEEE site.  I do see a conference presentation at https://asap.asu.edu/wp-content/uploads/2021/11/iccad17_asu_asap7_171115c.pdf

maliberty, 2023-01-08:
In general see https://asap.asu.edu/

closed 2023-11-02 by maliberty