The-OpenROAD-Project/asap7 #46: please open the road

anishsheikh, 2023-02-23:
please clear the roadblocks my friends from uni tried to acquire the calibre decks but stuck in potholes of ignorance. it would be great if you guys clear the roads from unneccesary roadblocks

maliberty, 2023-02-23:
It is explained on the front page:
[image]

Do you need something else?

anishsheikh, 2023-02-25:
well, we have already filled out it couple of times, I tried my personal email that didn't work, my friends tried their school emails, still nothing, some of them actually wrote to prof, Lawdence which, the same result. should i fill it out again with my personal email or school email because my school doesn't provide an .edu or school email my friends does.

vvashish, 2023-02-25:
> well, we have already filled out it couple of times, I tried my personal email that didn't work, my friends tried their school emails, still nothing, some of them actually wrote to prof, Lawdence [...]

Hi @anishsheikh. Sorry about this. It seems that we missed your email. Please email Prof. Clark again at lawrence.clark@asu.edu and cc me (vvashish@asu.edu). I will ensure that he does not miss it again. Be sure to use your University or work email address and to list your university/company affiliation.

anishsheikh, 2023-02-26:
thank you for help, my friend mailed from kiit.ac.in

closed 2023-03-07 by maliberty