The-OpenROAD-Project/asap7 #49: *.lib files cannot be called in yosys

Saazh, 2023-04-11:
We are trying to invoke abc inside of yosys. Since abc takes only one liberty file inside yosys at a time, i.e., we cannot pass multiple files as arguments, can we have a combined *.lib file for the ASAP7 or a solution to read all these *.lib files into the yosys?

Thank you ;)

maliberty, 2023-04-11:
I suggest you look at OpenROAD-flow-script.  The only real limitation is around dfflibmap but that only needs the .lib containing the ff.

rovinski, 2023-04-13:
IIRC there's also a limitation on file size in GitHub. If the .libs were assembled into a single file, the repo could not be hosted on GitHub.

nerdylye, 2023-07-26:
> I suggest you look at OpenROAD-flow-script. The only real limitation is around dfflibmap but that only needs the .lib containing the ff.

Hi, may I know where is the .lib file containing the ff that can be found in this github project? Thanks in advance

maliberty, 2023-07-26:
*SEQ* has the sequential cells

closed 2023-11-02 by maliberty