The-OpenROAD-Project/asap7 #56: Request for ASAP7 Calibre Decks

1864mp, 2023-08-10:
Hi I've requested a week ago calibre decks only I have had no reply - are the decks available to British universities? I can't see why not ARM was a British company afterall. Perhaps the link is dead?

maliberty, 2023-08-10:
I've tried sending a message to Prof. Clark on your behalf but that's all I can do.

1864mp, 2023-08-10:
> I've tried sending a message to Prof. Clark on your behalf but that's all I can do.

thanks, hopefully that'll get things going.

maliberty, 2023-08-10:
Please send me your email here or by DM on slack so I can pass it along.

1864mp, 2023-08-10:
hello that's mpanovic@ee.ucl.ac.uk 
thanks.

mrc-donato, 2023-09-08:
> Please send me your email here or by DM on slack so I can pass it along.

Hi, I've also been trying to access the calibre decks. Would you mind sending me a copy as well? Thanks!

maliberty, 2023-09-09:
Have you put in a request through https://asap.asu.edu/download/ ?

mrc-donato, 2023-09-09:
Yes, I submitted a request last week but haven't received a reply yet

maliberty, 2023-09-11:
Let me know you email here or an DM on slack if you prefer.

closed 2023-11-02 by maliberty