The-OpenROAD-Project/asap7 #58: Question related to M5, M6 spacing?

sakundu, 2023-08-26:
In the Post-Route-Opt database of one of our internal design on ASAP7, we noticed lots of spacing DRC. After further analysis we found that for M6 the [minimum spacing](https://github.com/The-OpenROAD-Project/asap7/blob/cd13de4e603913291ec6b8401ab63ec481178a5a/asap7sc6t_26/techlef_misc/asap7sc6t_tech_4x_210831.lef#L543) requirement is 0.128. But, the required spacing is 0.288 according to the [parallel run length spacing table](https://github.com/The-OpenROAD-Project/asap7/blob/cd13de4e603913291ec6b8401ab63ec481178a5a/asap7sc6t_26/techlef_misc/asap7sc6t_tech_4x_210831.lef#L562-L565). Which indicates that we always need to maintain 0.288 spacing between two M6 shapes. So, is this spacing rule correct?

Additionally, if we need a 0.288 spacing, is it acceptable to adjust the M6 layer pitch to 0.416 (combining 0.288 spacing and 0.128 width)?

maliberty, 2023-08-26:
In https://github.com/The-OpenROAD-Project/asap7/blob/master/asap7sc7p5t_28/techlef_misc/asap7_tech_4x_201209.lef it seems correct.  I'm guessing this is a bug.

maliberty, 2023-08-26:
It looks like Anton fixed the one above in https://github.com/The-OpenROAD-Project/asap7/commit/b86c7e9336427b85ba7eaf5eb64dcef53a85c0f9 - I guess the same fix is needed for the version you are using.

maliberty, 2023-08-26:
If you want to make a similar PR I'm fine with approving it.